TDPS-Mihotel/Mihotel #2: points to mention in report

LeoJhonSong, 2020-03-02:
Let's leave comments in this issue to collect points to write in report 👍
@TDPS-Mihotel/document

LeoJhonSong, 2020-03-02:
- our development strictly follows the V model

shuchang, 2020-03-02:
* We employ the **Gantt chart** to manage the project schedule
[image]

LeoJhonSong, 2020-03-10:
[Very good reference](https://bj.bcebos.com/cdstm-hyetecforthesmartcar-bucket/bb65363aa3ab442cb9732a7a95a943b0.docx) :satisfied:

LeoJhonSong, 2020-03-13:
Actually it is ridiculous that not even one report we saw has mentioned a single word about **teamwork** while this course is called `Team Design Project and Skills`. So I created a new issue #25 to collect teamwork contents we could talk about in our report.

LeoJhonSong, 2020-04-21:
We use git to  control the version of our code, hosting on github, which provide us with a lot of additional features and third party services, such as ZenHub.

LeoJhonSong, 2020-05-16:
我觉得应该说一下我们对信息的使用的准则: [关于如何使用传感器算是合理的](https://github.com/tdps-mihotel/mihotel/issues/55#issuecomment-628147856)

规定好这个才能谈后续的, 不然很多事情不知道是不是做过界了. 规定了这个才好说我们如何利用规则设计的我们的解决方案.

LeoJhonSong, 2020-05-16:
@TDPS-Mihotel 大家来集思广益啊 :muscle:

LeoJhonSong, 2020-05-21:
线上协作开发让我们有机会接触这种先设计, 仿真然后才进行实物测试的模式. (虽然我们没有最后一步) 在我们的讨论中可以发现当我们从零开始设计时这样做是最高效, 有方向性的:
- 给出项目的定位
- 定下项目的规格, 技术指标
- 根据这个总的指标分解为各部分的指标
- 根据指标选型, 查找参考资料, 给出可能解决方案
- 测试解决方案, 调整解决方案

实际上这正是个V-model的具体例子. 如果我们不给自己设定技术指标那么在开发过程中随时有可能产生不合理的设计, 不必要的困扰. 退一步讲也会导致团队内沟通不畅, 信息混乱. 而如果不根据技术指标来设计解决方案那很容易找不到切入点, 迷失方向, 或者做出看似符合要求实则无效的解决方案, 很容易浪费时间.

closed 2020-06-03 by LeoJhonSong