TDPS-Mihotel/Mihotel #11: Reimbursement process requirement

LeoJhonSong, 2020-03-06:
@TDPS-Mihotel/project-manager Please

- [ ] set up:
1. our reimbursement process
2. what information is neededd when they ask for budget from you

- [ ] update those to repo Mihotel
:bulb: our budget should always updated there.

In that way every one are free to look for devices or circuits to buy ​:satisfied:​

LiamBishop, 2020-03-10:
购买方名称: 电子科技大学
纳税人识别号:  121000004507193117
地址: 四川省成都市高新区(西区)西源大道2006号电子科技大学清水河校区
电话： 可填采购人电话
开户行及账号: 中国建设银行成都新鸿支行51001908608059000888 (非科研业务)

LiamBishop, 2020-03-10:
The sum of money on the invoice should be matched to the trabsaction record of Alipay. In other word, the freight should be contained in the total price.

closed 2020-03-14 by LiamBishop