TDPS-Mihotel/Mihotel #16: Start visual development

LeoJhonSong, 2020-03-06:
@TDPS-Mihotel/Visual Please create your own repo **from your team's page** and start to look for method to reach your tasks.

:bulb: since there are quite a lot of tools and method to deal with images, I suggest you **just browse first** and update useful reference to your repo to share ideas with each other (or simply chat by QQ)

LeoJhonSong, 2020-03-09:
This may helps you
[树莓派上利用Tensorflow实现小车的自动驾驶](https://my.oschina.net/u/3858986/blog/2875027)

shuchang, 2020-03-17:
Some DL methods to do path recognition:
* Supervised Learning:
  * [Marking the center point](https://becominghuman.ai/autonomous-racing-robot-with-an-arduino-a-raspberry-pi-and-a-pi-camera-3e72819e1e63) ([中文翻译](https://cloud.tencent.com/developer/news/73131))
  * [Reference 1 for some state of the art](https://zhuanlan.zhihu.com/p/37801090)
  * [Reference 2 for some state of the art](https://www.jiqizhixin.com/articles/2018-06-04-17)

**Strengths**: Given the labels, we can include images in extreme environmental situations to enhance the robustness of the trained model
**Weaknesses**: We need to give labels to the training sets manually

* Unsupervised Learning:
  [Unsupervised-Segmentation](https://kanezaki.github.io/pytorch-unsupervised-segmentation/ICASSP2018_kanezaki.pdf) ([中文翻译](https://zhuanlan.zhihu.com/p/68528056))
  [FCAN](https://arxiv.org/abs/1804.08286) ([中文翻译](https://blog.csdn.net/longxinghaofeng/article/details/83993614))
  [Double DIP](https://arxiv.org/abs/1812.00467) ([中文翻译](https://www.bilibili.com/read/cv2802966/))

**Strengths**: No need for labeling
**Weaknesses**: To obtain good clustering results during training, photos should have clear boundaries. However, during the test, the quality of photos might suffer from environmental factors, such as sunlight, shadow and so on

In general, we hope to use the DL methods to achieve two goals:
* Obtain a pre-trained model for the path recognition
* Learn features that traditional image processing methods fail to learn

closed 2020-04-08 by LeoJhonSong