TDPS-Mihotel/Mihotel #20: Multi-thread in python

LeoJhonSong, 2020-03-06:
As we want our rover to move around smoothly, multi-thread in needed.

For example, when recongnizing the route, we need a thread to do **image processing**, another thread to control the movement, and maybe some more threads to get the sensors running. If we use only one thread, the rover need to judge if is getting off the route, then move for some distance, then again do the judgement... It show out that the rover can not move around smoothly.

Let's try out a template which allows up to 3 thread first, then contact with the control group to see how many threads we actually need :+1:

LeoJhonSong, 2020-03-06:
@Laince20 do you want to do this?

Laince20, 2020-03-07:
i think it is OK

Laince20, 2020-03-10:
https://www.liaoxuefeng.com/wiki/1016959663602400/1017628290184064

> python 多进程

LeoJhonSong, 2020-03-10:
Sorry for the late reply... :cry: 

 Alright this is my first time to check out [the difference between multi-threading and multi-process, and which one to choose](https://www.jianshu.com/p/a69dec87e646). I think your choice of using multi-process is right based on its suggestion:
> 如果程序是属于 CPU 密集型，建议使用多进程。而多线程就更适合应用于 I/O 密集型程序。

But it seems communication between processes could be a issue, which we must need, for example the dicision (决策) process will check information from other processes frequently.

In addition, as we can simply start a process by attaching a function, we can easily make functions into modules or methods of a class.

So I suggest you do these two work:
- try to develop a tiny program to test **communication** between processes
- learn about **class**, **module** in python

I will update what we may need to be care of in communication between processes later , also some recommanded useful materials about class, module.

commit ee6524aa (2020-04-21): feat(system): update initial system example

create a system running three processes, that is, movement process,
sensor  process and decision process
This system allows processes to communicate with each other

Closes #20
commit 375d442b (2020-04-26): feat(system): update initial system example

create a system running three processes, that is, movement process,
sensor  process and decision process
This system allows processes to communicate with each other

Closes #20
closed 2020-04-21 by LeoJhonSong