TDPS-Mihotel/Mihotel #7: Let's select a ASCII art logo😏

LeoJhonSong, 2020-03-06:
It will be used in ssh connection and also transmitted message in p2t3.

 :link: find a "Mihotel" in fancy font [here](http://patorjk.com/software/taag/#p=display&f=Acrobatic&t=Mihotel)

Let's leave comments to select one :+1:

some good one I found:

- 3D Diagonal

[image]

- Star Wars

[image]

- Roman

[image]

- Reverse

[image]

- alpha

[image]

LeoJhonSong, 2020-03-06:
Font Name: ANSI Shadow

[image]

Prince-JIAN, 2020-03-07:
Calvin S[image]

LeoJhonSong, 2020-04-08:
OK, I'll pick my modified version of **Big Money-ne**, code is here:

```
     /██      /██ /██ /██                   /██               /██
    | ███    /███|__/| ██                  | ██              | ██
    | ████  /████ /██| ███████   /██████  /██████    /██████ | ██
    | ██ ██/██ ██| ██| ██__  ██ /██__  ██|_  ██_/   /██__  ██| ██
    | ██  ███| ██| ██| ██  \ ██| ██  \ ██  | ██    | ████████| ██
    | ██\  █ | ██| ██| ██  | ██| ██  | ██  | ██ /██| ██_____/| ██
    | ██ \/  | ██| ██| ██  | ██|  ██████/  |  ████/|  ███████| ██
    |__/     |__/|__/|__/  |__/ \______/    \___/   \_______/|__/

 ```
It taste good in terminal
[image]

closed 2020-04-08 by LeoJhonSong