TDPS-Mihotel/Mihotel #9: create repo for chassis

LeoJhonSong, 2020-03-06:
@TDPS-Mihotel/Chassis Please create a repo **from your team page** (therefore it belongs to your team) for your chassis development.

Items this repo could be:

- 3D model files of the whole rover
- documents of chassis development so everyone can know your work
- code of chassis
- main references of each device provided by taobao

LeoJhonSong, 2020-03-09:
Observing how other chassis arrange all the components may helps:+1:

[image]
[image]
[image]
[image]

Howard2503, 2020-03-17:
chassis model modification

closed 2020-03-19 by LeoJhonSong