TDPS-Mihotel/Mihotel #22: Keil Set up and STM32 F4 Firmware Library add.

allensted, 2020-03-10:
please finish Setting up Keil5 , Firmware Library, and successfully compile the code that I send to you in QQ group.

I hope we can finish the setting up task before weekend, so we can move on and learn some new knowledge.

LeoJhonSong, 2020-03-10:
I recommend you guys open an **extra repo** to do this (as I mentioned in #9) since the code for STM32 and the code for RPi are very isolated from each other.

LeoJhonSong, 2020-03-10:
@allensted I think you have assigned the wrong person? `Laince20` is 许翰鹏's ID

Laince20, 2020-03-10:
we have talk about this mission before. It is the basics of control group, so there is no problem

LeoJhonSong, 2020-03-10:
> we have talk about this mission before. It is the basics of control group, so there is no problem

I mad a stupid mistake on the name "control group", sorry about that. I will talk with allensted later

closed 2020-03-15 by Prince-JIAN