TDPS-Mihotel/Mihotel #26: Start decision

Prince-JIAN, 2020-03-15:
Collect information and discuss the overall process.

Laince20, 2020-03-15:
[image]

LeoJhonSong, 2020-03-16:
首先有几点需要注意一下:

- 我们的底盘因为成本, 能力等原因无法做到很精确, 然后因为目前使用的底盘是靠**舵机转向的**, 无法像b站视频的四轮驱动小车那样灵活原地转向. 所以**目前不能把转向角度作为一个控制指令**
- 最后我们决定使用的底盘不是步进电机的, 而是直流电机的. 不过你们想要测距的话可以通过**码盘**来实现

然后我对具体每点说下我的看法.

> case2: 若路线不为直线, 停下处理图像信息, 判断是否要转弯.

这里我们不应需要停下来. 计算机不是人脑, 处理速度很快而且可以多进程操作. 如果因为图像识别速度太慢所以需要停下来, 那是视觉组的锅:smirk:

> 若出现偏差, 需根据实时图像去纠正偏差

这个实在是一个很好的方法! 在我们通过惯性导航无法精确导航, 而我们又可以设置两个信标的情况下这是很好的定位方式. 我的设想是这样: 在桥或者桥前贴一条颜色, 形状都十分利于识别的贴纸 (比如绿色L字), 然后通过识别这个形状在图像中是不是正的来判断是否已经转到对准了桥, 然后就回正轮子. 这样我们就**不需要依据转90度这个很难准确达到的依据来判断是否转正了**. (更何况桥其实也不一定就能摆得和路完全成90度)

[image]

> 小车下坡后, 到达信标T2 (8位置), 然后左转90度, 准备过门

我又仔细看了一下课程说明, 实际上**只有p1t1是需要巡线的**, p1t3只是要求**通过门**, 因此此处或许不必严格转90度, 甚至可以在下桥后就开始转弯.

> 若捕捉到色块, 向左转45度, 向远方捕捉相同色块 

这个地方其实我们应该是可以直接看到其他色块的, 也就是说**可以直接奔着目标色块去**. 因为视觉组的前视摄像头不可能是水平向前的, 必定有一定**俯视角**, 通过底盘组的建模以及摄像头卖家提供的视场角参数我们可以得知摄像头能看到的**远近范围** (当然这个不会完全准, 所以 底盘组给摄像头设计的固定机构最好能便于调整摄像头角度)

> 捕捉色块后调整角度直行...

从色块处到投食处的方案我建议你们完全交给传感组+视觉组的同学来做, 相信他们会找到更好的方式:grin:

> 然后原地识别第二个信标

emmm在从色块到投食处的过程中可能就需要两个信标了... 这点再商议吧:+1:

LeoJhonSong, 2020-03-16:
另外把issue标题改成更准确的, 比如 `Split project into functional tasks`

LeoJhonSong, 2020-03-19:
不好意思之前光顾着研究你们的方案忘了说你们的步子迈大了, 一开始的任务分解不必做到这么细节.:joy:  我来开个头写个P1吧:smirk:

## Patio1

1. **巡线走**

   :exclamation: 实际上从handbook上可以看出其实t1在到桥前面有一段就已经结束了, 这意味着我们可以在到桥前就开始转弯

2. 在7号位置附近**获得转弯信号**并**调整方向**以对准桥

3. **直走** (通过桥)

   我觉得一旦对准了桥, 按照桥宽`45cm`, 桥长`220cm`, 车宽`14.8cm`, 只走两米多我感觉不大可能偏差近**15cm**, 因此可以只在上桥前进行方向校准, 之后就直走

4. **获得转弯信号**并**调整方向**对准门/回到线路

   :exclamation:  t1之后并不要求巡线, 但因为门就横跨线路因此可以利用这点来校准方向.

5. **直走** (通过门)

:bulb: 一些可能方案

两次获得转弯信号都可以考虑使用[TPDS Team11-18秒处](https://www.bilibili.com/video/av67433575?from=search&seid=7007344396073135097)的方案: 将一条黑色带贴得对准桥的方向, 能够**同时给出转弯信号和作为方向调整的参考线**, 实在精彩. #33 就是在解决这两个信号问题 @wb05025 来康康这个

:exclamation: 但是感觉这个色带应该是有一定限制的, 不然直接一条拉满全场. 如果考虑采用这一方案应当询问wasim态度.

[image]

而第二次需要转弯信号处还可以考虑使用**超声波传感器**, 因此根据示意图和[视频53秒处](https://www.bilibili.com/video/av53595153?from=search&seid=7007344396073135097)可以看出`8位置`前方不远就是墙

[image]

LeoJhonSong, 2020-03-19:
关于P2的一些可能方案:

[这个组 (1分11秒起)](https://www.bilibili.com/video/av53293727?from=search&seid=7007344396073135097) 似乎是通过巡**湖边上这个线**来从色块处到达投食处的:joy:

[image]

[image]

如果真的能控制住不掉到湖里这简直是暴力破解:joy:

[image]

而从这个角度看如果色带黑科技真的能用这里同样可以通过一条色带矫正方向直接直走到p2t3结束处

Prince-JIAN, 2020-03-22:
[Task1.pdf](https://github.com/TDPS-Mihotel/Mihotel/files/4364832/Task1.pdf)
[Task2.pdf](https://github.com/TDPS-Mihotel/Mihotel/files/4364833/Task2.pdf)
Patio 1&2 的基本任务拆分 需视觉与传感组对于细节再进行明确 提供可行性方案

closed 2020-05-13 by Prince-JIAN