TDPS-Mihotel/Mihotel #31: chassis model modification

Howard2503, 2020-03-17:


Howard2503, 2020-03-17:
- placement of RPi and STM32
- placement of sencors(if any)
- design feeding part

LeoJhonSong, 2020-03-18:
These two designs may help you. (It seems that there are **3D models** in their github repo)

- [donkeycar](https://github.com/autorope/donkeycar)

[image]

- [RacingRobot](https://github.com/sergionr2/RacingRobot)

[image]

LeoJhonSong, 2020-03-19:
Just as the two examples above, I suggest  the holder you design could allow to some extent **adjustability**. For example adjust the high of camera, the angle of camera, since even we can get a reasonable configuration by simulation in SolidWorks, but the components may not reach that accuracy we thought. Therefore, to some extent adjustability could be really helpful.

LeoJhonSong, 2020-03-19:
> * placement of RPi and STM32
> * placement of sencors(if any)
> * design feeding part

You forgot a area for the battery placement :smirk: (it could be a bucket or just fix it by a few strappings)

LeoJhonSong, 2020-03-24:
Here are some useful 3D models:
- [Raspberrypi 4B](https://grabcad.com/library/raspberry-pi-4-model-b-1)
- [Raspberrypi 3B (very beatiful)](https://grabcad.com/library/raspberry-pi-3-model-b-reference-design-solidworks-cad-raspberry-pi-raspberrypi-rpi-1)

Also you can find more 3D models on this powerful site.
- [Raspberry Pi Camera Module](https://grabcad.com/library/raspberry-pi-camera-4)

closed 2020-04-08 by Howard2503