TDPS-Mihotel/Mihotel #32: Visual-Path following course detection algorithm development

wb05025, 2020-03-19:
Patio1-task1: From the start, follow the path to the bridge
Purpose: to detect the path to be followed, its direction and return correction angle. 
We will try both machine learning methods and regular image processing methods.

The goal is to detect the course with no error in general direction and error as small as possible in specific angle.

Difficulty: Unclean surface disturbance\shadow of the car\more accurate calculation in curve segments.

First try by regular IP methods:
[image]!
[path2.JPG](https://images.zenhubusercontent.com/5e70d3478ec11b0fa1ec205a/00d1fffa-a7f1-4c98-8c4f-22a4005acf44)!
[path3.JPG](https://images.zenhubusercontent.com/5e70d3478ec11b0fa1ec205a/72d76a55-4a20-4761-8ba8-1a03efd543a7)
[image]
[image]

MATLAB script:
[https://app.zenhub.com/files/243200095/8d2589e8-6ff7-4e8f-9f14-b31649803dba/download](https://app.zenhub.com/files/243200095/8d2589e8-6ff7-4e8f-9f14-b31649803dba/download)

wb05025, 2020-03-19:
[image]
[image]

wb05025, 2020-03-19:
New MATLAB script with minor fix and annotation
[https://app.zenhub.com/files/243200095/f1dee83a-ca12-4726-83c5-39c7944e7ce7/download](https://app.zenhub.com/files/243200095/f1dee83a-ca12-4726-83c5-39c7944e7ce7/download)

LeoJhonSong, 2020-03-19:
I suggest you create a more specific epic for this **path recognition** topic

closed 2020-04-21 by LeoJhonSong