TDPS-Mihotel/Mihotel #35: car model and movement

Howard2503, 2020-04-08:


LeoJhonSong, 2020-04-21:
please update the status of this issue.

LeoJhonSong, 2020-04-22:
[image]

It seems [SolidWorks could export VMRL 2.0 file](https://help.solidworks.com/2017/english/SolidWorks/sldworks/HIDD_EXPORT_OPTIONS_VRML.htm).
You may have a try.

Howard2503, 2020-04-22:
the model we used seem to have to many components which require me to assign their properties seperately. And scaling in webots cause unexpected problems. I am looking for a solution

LeoJhonSong, 2020-04-22:
[This](https://jingyan.baidu.com/article/d169e1864cc54e436611d832.html) may helps you. You can merge most components into, say just body and four wheels.

Howard2503, 2020-04-22:
thanks I will try this

closed 2020-05-16 by LeoJhonSong