TDPS-Mihotel/Mihotel #36: Webots visual sensors and recognition algorithms

shuchang, 2020-04-08:
Let's start to familiarize ourselves with various visual sensors in Webots and provides corresponding path recognition algorithms.

shuchang, 2020-04-14:
The path recognition algorithm task is removed from this issue and we create a new issue for it. In this issue, we mainly focus on the usage of various visual sensors, like the camera, the distance sensor, the Lidar, and so on. The current goals are categorized into 2 types.
* master the basic operations of these sensors
* dig out some advanced operations of these sensors so that they can work more intelligently

In short, to obtain better path recognition algorithms, one good way is to deeply go through the sensors we have right now.

HandAdam, 2020-04-16:
xxx

HandAdam, 2020-04-16:
Usage of the distance sensor:

- Firstly, add the **"disatncesensor"** to the children part of the robot.

- By modifiying the **"translation"** and **"rotation"** property, we could change the relative position between the robot and the sensor. By modifying **"rotation"** property, we could change the direction towards which the sensor detct. 

- Noticing that, the sensor always detect along the **_positive dirction of the x-axis_**.

- Changing the **"name"** property of the sensor, and this will be the name that we will use in the ### python code.

HandAdam, 2020-04-16:
Usage of the Opencv package in the Webot:

- Firstly, simply import the package
`import numpy as np
import cv2`

- Applying `.getImageArray()`, we could get a list that represent the picture.

- By applying the next sveral line of code, the picture could be process by the build in function of opencv
`image = np.array(camera.getImageArray(),dtype="uint8")
r,g,b=cv2.split(image)
image=cv2.merge((b,g,r))
`

1. The default type of the picture get by the Webot camera is **'int32'**, which could not be processed by the opencv. In that case, we need to set `dtype="uint8"` so that the opencv could process without error.
2. The channel need to be changed.

LeoJhonSong, 2020-04-21:
> * Firstly, simply import the package
>   `import numpy as np import cv2`

I do not suggest to import more than one packages in a single line.

```python
import numpy as np
import cv2
```
This could be a better way.

LeoJhonSong, 2020-04-21:
@HandAdam I suggest you record all above content to our document. That could be really helpful. And as you finish this issue please close it :+1:

HandAdam, 2020-04-28:
Usage of the Compass Sensor:

- Add a compass as the children of the robot. Change it name to what you want or the default "compass"

- in the conntroller.py code:
```
from controller import Compass
Compass1=Compass("compass")
Compass1.enable(1)
print(Compass1.getValues())
```
could dirctly return a list with shape=[3], dtype=float64. The return shows the dirction where the z-axis points to.
float 64

HandAdam, 2020-04-30:
Usage of the Compass Sensor:

- Add a compass as the children of the robot. Change it name to what you want or the default "compass"

- in the conntroller.py code:

```
from controller import Accelerometer
Accelerometer1=Accelerometer("accelerometer")
Accelerometer1.enable(1)
print(Accelerometer1.getValues())
```
could dirctly return a list with shape=[3], dtype=float64. The return shows the acceleration along each axis.

- The return of **y-axis** is  the real acceleration with a gravity, g, added

HandAdam, 2020-04-30:
Usage of the GPS Sensor:

- Add a GPS as the children of the robot. Change it name to what you want or the default "gps"

- In the conntroller.py code:
```
from controller import GPS
GPS1=GPS("gps")
GPS1.enable(1)
print(GPS1.getValues())
```

- could dirctly return a list with shape=[3], dtype=float64. The return shows the absolute position of the sensor along each axis.

closed 2020-04-30 by HandAdam