TDPS-Mihotel/Mihotel #55: Decision & Visual Group discussion

Prince-JIAN, 2020-04-28:


Prince-JIAN, 2020-05-13:
[image]

LeoJhonSong, 2020-05-13:
# 关于如何使用传感器算是合理的

我今天边听课边跟着讨论脑子乱乱的, 因为讨论期间我们多次为"这样用传感器算不算犯规"而困扰, 我列一下自己关于合理使用传感器更具体的想法供大家参考.

我觉得摄像头, 距离传感器, 罗盘, GPS等传感器的使用实际上都是合理的, 即便是将这个任务套入真实世界这些传感器的使用也是必不可少的, 不然这个任务不可能只靠两个信标就完成.

### 关于传感器的精度

虽然目前我们使用的GPS的精度(**完全精准**)比真实世界民用级GPS精度(**10米级**)高很多, 但这是因为我们的车小, 路线也细. 如果对标AGV常用的基于UWB的室内定位系统(普遍达到**10厘米级**), 会发现其实精度差不太多, 是合理的:smirk:

这里唯一精度有点bug的是**罗盘**, 不过没必要去设置罗盘的精度. 我认为我们的算法不会需要依赖罗盘的精度, 这个bug可以忽略.

### 关于什么信息是我们知道的

为了不陷入"不知道这样使用传感器算不算作弊"的困境, 我建议我们在设计方案时将这个patio作为一个真实世界来看待, **以有基于UWB室内定位系统的AGV的条件来考虑**, 暂时不以往届TDPS的条件来考虑, 那样未免有些难为自己, 留作后面有闲工夫做.

我来举一些例子.

我觉得我们可以将老师的这张图作为地图, 认为**地图中给出的数据都是可以使用的**.
[image]

比如通过z轴坐标小于0来判断小车现在在上半部分, 即**已经过了桥**是合理的.
比如通过桥后向右走20m来让小车的x坐标和门的x坐标相近是合理的.

而这张地图**没有给出具体数值**的信息我们认为是信息不完全的, 只能现场识别. 其实这个很好判断, 在你看到环境后, **仍然需要问环境组才能知道的信息就是我们不应该知道的**.

比如我们**不知道路线是不是平行于x轴的**. 因此在从橙色盒子到桥的过程中不能直接直行, 应该根据GPS/罗盘数据来确保沿x轴方向前进, 或者说通过GPS关注z轴坐标来确保不会掉进池塘就可以.

Prince-JIAN, 2020-05-19:
5.19
[image]

LeoJhonSong, 2020-05-19:
[image]
为了能确认每一个步骤我拆得比较细, 可能有点乱. 似乎黄色部分是目前仅有的还没定下来的部分?
@TDPS-Mihotel/visual @TDPS-Mihotel/sensor 标黄部分有什么好方案吗.

(我在这备份个这个图的[源文件](https://github.com/TDPS-Mihotel/Mihotel/files/4652715/CFG.zip))

HandAdam, 2020-05-20:
建议的黄色部分，只要所在位置的z坐标小于一个值，就当成是已经过了桥了

LeoJhonSong, 2020-05-20:
> 建议的黄色部分，只要所在位置的z坐标小于一个值，就当成是已经过了桥了

确实是一个方法, 但这要求我们知道河的边界. 这是地图里没有给的, 感觉这么用有点破戒. 备选吧.

LeoJhonSong, 2020-05-20:
哦哦这套方案里好像还没用信标, @HandAdam 要不你们在这部分用一个信标得了.

closed 2020-05-25 by LeoJhonSong