TDPS-Mihotel/Mihotel #63: image captured are not in right direction

LeoJhonSong, 2020-05-12:
摄像头实际拍摄到的画面是下图这样的
[image]
但保存出来的图片是这样的
[image]

这个方向没对, 需要你修一下:+1:

另外右边的estimate point记得设置一下.

LeoJhonSong, 2020-05-12:
另外原本我的想法是让他保存在这里 (如下图) 的, 结果疏忽了, 写成保存在`webots/controller/mihotel_rover/test/camera/0/`下面了... 我在[]()里把这个路径改过来了.

另外我建议可以把这个函数再封装好一些, 比如:
```python
    def capture(self, index):
        '''
        capture images from camera to test/camera/.
        '''
        capture_path = '../../../test/camera/' + str(index)
        if not os.path.exists(capture_path):
            os.makedirs(capture_path)
        image = self.get_image(index)
        cv2.imwrite(capture_path + time.strftime("%Y-%m-%d-%H-%M-%S", time.localtime()) + '.png', image)
        info('Captured!')
```
这样可以把对应相机的图片存到对应index的文件夹下, 调用也容易一些, 直接`self.capture(3)`就可以.

closed 2020-05-13 by HandAdam

HandAdam, 2020-06-05:
[image]!

[2.png](https://images.zenhubusercontent.com/5e5e045cbf668358438d1902/198c3a07-9dd6-40dd-84a4-02d2fbf08ca5)

commit 627cc357 (2020-05-13): Update detection.py

close #63
commit 608519a3 (2020-05-16): Update detection.py

close #63
commit e670a421 (2020-05-16): Update detection.py

close #63

Former-commit-id: bee4c66f29b37d484f225f68e99d4b0c5007c56b