TDPS-Mihotel/Mihotel #72: give the value range of colors we concerned

shuchang, 2020-05-16:
The color detection module should be able to distinguish the colors of the following objects. Here I make a rough division.
* blue: the pond, the river, and the sky, the color box
* red: the color box
* green: the color box
* orange: the orange box
* yellowish-brown: the sand

LeoJhonSong, 2020-05-16:
No, I think we just have to **focus on our interested colors**, pixels in other color should be simply filtered, for example the color of pond, river or sky is now not our interest.

Our interest colors now are:
- **red** of colored box and line
[image]
- **yellow** of colored box and line
[image]
- **purple** of colored box and line
[image]
- **orange** of orange box
[image]

So what you need to do is operating the source image with the masks you designed, to see if there are pixels of these colors. If there is, we get a target.

:exclamation: Note that Wasim did not specify the exact color values :smirk: so if the yellow line on sand ground is difficult to recognize, you could choose a **specific yellow** which makes it easier. Or, if it is still too difficult, ask the environment group to place a white plane under the lines in task3.

shuchang, 2020-05-16:
Oops, I was thinking the three colors in Task 5 are red, green, blue. 
If I get it right, I need to specify the range of these four colors and change the program from finding the principal color into giving a signal once we find a color we want. No problem!

LeoJhonSong, 2020-05-16:
Exactly.

LeoJhonSong, 2020-05-16:
[Here](https://github.com/jrosebr1/imutils/blob/master/bin/range-detector) is a script which helps you to find the threshold with **GUI** and **real-time**.
Just copy the content of this script to a file enjoy!
An example to run this script:
```shell
python range-detector.py -f HSV --image ./test.png
```

LeoJhonSong, 2020-05-16:
> I need to specify the range of these four colors and change the program from finding the principal color into giving a signal once we find a color we want.

I suppose [this](https://www.pyimagesearch.com/2014/08/04/opencv-python-color-detection/) could help you to do that.

shuchang, 2020-05-22:
It seems that fine-tuning the threshold with GUI does not bring about satisfactory results.  I can only find a broad range with this approach,  which causes other colors to be recognized as target color as well. 
I suggest using the Snipaste to find the accurate R, G, B values of the target color and set the range with a little tolerance.

LeoJhonSong, 2020-05-22:
> It seems that fine-tuning the threshold with GUI does not bring about satisfactory results. I can only find a broad range with this approach, which causes other colors to be recognized as target color as well.

Noise reduction is really an art. You may have to search a lot to figure out some filter method that helps impproving you recognition. Best Wishes. :muscle:

LeoJhonSong, 2020-05-24:
要注意因为阿韩的算法是将颜色较深的像素作为路面, 因此你应当将要滤掉的像素值设为白色

closed 2020-05-26 by LeoJhonSong